Siedmiu/Hexapod #19: Walking Algorythm Blending

Siedmiu, 2025-03-13:
@Bonk03 and @JuliaHry have already created gait cycles for the robot, which are our building blocks for more diverse movement.
One of the most complicated problems in our movement kinematics is blending them together. You have to figure out how to transition between the movements (this is the easier part), and how to make micro adjustment to the walk cycle to be able to follow curved lines.
Until #MISSING_ISSUE is complete use 0.1*x*sin(0.05x) as a test path.

Bonk03, 2025-03-23:
I was able to count the trajectory of the leg movement it needs to make to get from point A to B by passing through the selected peak point. I tried to work on a paraboloid intersected by a plane common to the points, nothing made sense to me, there were cases in which what I calculated did not make sense. Finally, I came up with the idea of parametrizing the function of each variable x, y, z so that when opbuting one variable does not affect the other. This works exactly as I planned. The code is uploaded to moving_codes, 3D_points_of_movement.py. I think we will be able to use this once we have determined the points where the legs should hit in order for the spider to move along the curve.

Siedmiu, 2025-03-27:
I tried, but didn't succeed. Implementation of ripple gate might be impossible for curved paths (maybe smooth ones will work). Wave gate might be the best for it because it minimizes the chance of leg colisions.
There are 2 approaches I tested, add micro adjustments of goal leg positions based on the path, or a better approach to transform the robot coordinate system so that it is alligned with tangent and normal of the curved path at each sampling point.
There is also a problem with keeping the static legs "glued" to the ground as the robot will pass through many singularity points.
I found a path planning chart for this in a research paper:
1. Pre defined required body path 
2.finds foots 1-6 start positions 
3. Calculate each leg stance phase path points to achieve the required body path 
4. Check if the stability at each point along the path. NO -> go to 7, YES -> go to 5 
5. Calculate each leg inverse kinematics 
6. Check the legs kinematics limitations NO -> go to 7, YES go to 3 
7. Each leg joints Path

Bonk03, 2025-03-31:
I calculated how leg has to move to rotate hexapod and move it at the same time. Code is in project. I only checked it on one example. I will see if other examples works.

Bonk03, 2025-04-02:
Other example:

P = (0,0)
delta x = 1
delta y = -2
N = (-2, 4)
A = N + (-6, 3)

[image]
geogebra results:

h = x = 4.8668850910106
i = -y => y = -6.8397869778364

python results:
x = 4.8668850910105546 
y = -6.83978697783636

Bonk03, 2025-04-02:
I think this will be the last example that I uppload here, every option works fine.

P = (0,0)
delta x = -3
delta y = -1
N = (5, 0)
A = N + (3, 0)
alfa = 45 deg

[image]
geogebra results:

h = x = 3.4852813742386
i = -y => y = -7.0710678118655

python results:
x = 3.4852813742385718 
y = -7.0710678118654755

Siedmiu, 2025-05-31:
Extensive work regarding the algorithms and their parametrisation has been conducted. The results are yet to be tested thoroughly on the new physical implementation. 
Simulated robots can prepare their leg positions for the walk cycles.

This issue is potentially going to be revisited in phase 2 of the project where smooth pathing will be researched.

Current implementation is satisfactory and the issue is marked as done and closed.

Bonk03, 2025-06-02:
my commits for this issue:
739441f
f52e8d3

JuliaHry, 2025-06-02:
My commits for this issue:
1349058202997ed431ca868a2fa2bd06480c6453
da03ac1539995f1b148094c944f2b53d9ae5ee28