Siedmiu/Hexapod #35: documentation

FlyBozon, 2025-05-31:
- [ ] user manual
- [ ] software documentation
- [ ] hardware documentation

Siedmiu, 2025-06-02:
A general plan for the documentation as well as hardware documentation is ready to be documented

Siedmiu, 2025-06-06:
> A general plan for the documentation as well as hardware documentation is ready to be documented
What was I on?

Most of the previously unrecorded issues were added as retrospective issues and have relevant commits attached to them.
Most of the code is still in state of dissaray when it comes to comments and documentation.
I am actively working on mechanical documentation that explains design decisions.
The hardware documentation needs to be gathered in one place, pdf documment of sorts, this shouldn't take more than an hour.

Siedmiu, 2025-06-11:
We have a main Overleaf Latex file where the documentation for all of the features will be combined