Siedmiu/Hexapod #36: printing 3D parts

Bonk03, 2025-05-31:
printing parts, some of us are doing it in proto some of us are doing it at home

MikolajGozdzielewski, 2025-05-31:
27.05.2025
Mikołaj Goździelewski 15:00-16:00, Bartłomiej Bąk 15:00-16:00, Bartosz Babiński 16:00-17:00
28.05.2025 
Mikołaj Goździelewski 10:00-14:00, 15:00-16:00, Bartłomiej Bąk 10:00-14:00, 15:00-16:00
Wstawiliśmy druki na noc, nie przetrwało
29.05.2025
Mikołaj Goździelewski 09:00-12:00, 15:00-20:00, Bartłomiej Bąk 09:00-12:00, 15:00-20:00
Wstawiliśmy druki na noc, 1/3 wydruków nie przetrwała
30.05.2025
Bartłomiej Bąk 09:00-13:00, Mikołaj Goździelewski 12:00-16:00
Wstawiliśmy druki na noc, 1 całkowicie nieudany, z drugiego 1/3 elementów do wywalenia
31.05.2025
Mikołaj Goździelewski 09:00-19:00, Bartłomiej Bąk 09:00-19:00, Mariia Krykun 09:00-19:00

closed 2025-06-02 by Siedmiu

Bonk03, 2025-06-02:
fc36e95, photo of progress

Bonk03, 2025-06-02:
17bac21: weights of each element (without platform)