Siedmiu/Hexapod #47: move body with constant legs possitions

FlyBozon, 2025-06-02:
- [ ] implement some logic to move body relating to foots positions (currently we have logic that u can move legs relating to the platform, so it must be changed for our current goal or find some hack for that)
- [ ] implement visualization in python - _optional_
- [ ] implementation in Gazebo

Siedmiu, 2025-06-02:
What exactly are we planing to do here, and why is it required?

I don't think that it is possible to move the platform horizontally with big enough volume of movement without changing the position of the feet.
Positioning the platform UP and DOWN shouldn't be difficult to calculate since it only requires moving the joins in such a way that keeps their attachment point on one vertical axis. This shouldn't be difficult to calculate, or might even already be solved with the inverse kinematics equations.
Stick-bug side to side movements shouldn't be an issue either, but I don't see a reason for them.
Back and forth movements are interesting... I don't think we would be able to make long movements and avoid singularity points, but short ones may be possible. This could either be applied to readjust the position (especially with combination of 2 previous points) or intertwined with gait cycles to make the robot move faster - but very difficult.

There is another approach if our goal is to reach certain platform positions. We can try to estimate the best leg positions for a given platform position, but this is also difficult. I have already tried to implement that as a solution for curved path walking, where the platform was kept tangent to a given path. There is limited research online on how to do it, and my attempts were unsuccessful

[image]

FlyBozon, 2025-06-02:
@Siedmiu answering your question, that issue would be needed for uphill walking, to adjust platform position to compensate gravity impact. Idk if anyone else need that, but we can use it later to make robot dance, as once was told at the lecture - every robot should be able to dance (maybe it was told rather like - for some reason every robot can dance, but cannot do what they were created to, but nevertheless :D)

FlyBozon, 2025-06-24:
https://github.com/user-attachments/assets/f9501619-1c2f-43d9-ae1b-42393d639659

closed 2025-06-24 by FlyBozon